hydrabus/hydrafw #11: nfc mode locked to Mifare

doegox, 2015-03-26:
> nfc
NFC> show
Protocol: MIFARE (ISO14443A)
NFC> vicinity
NFC> show
Protocol: MIFARE (ISO14443A)

In principle mode should start as NONE but it's already = MIFARE and any attempt to change it fails.

I tried to dig it a bit, if I monitor "vicinity":
                case T_VICINITY:
+                       t += show(con, p);
                        proto->dev_mode = NFC_MODE_VICINITY;
+                       t += show(con, p);
                        break;
 I can see the mode changed to vicinity but there should be sth somewhere else that resets it back to MIFARE before the next command gets executed...

bvernoux, 2015-03-26:
Yes correct, I reproduced it on latest FW release (v0.5-beta-1-g4897d50), It is a regression I will fix it, thanks for the bug report.

bvernoux, 2015-03-28:
It is now fixed in trunk

closed 2015-03-28 by bvernoux