hydrabus/hydrafw #12: Nand support

relax79, 2015-04-23:
Hi bvernoux
Can you add firmware later x8 and x16 nand support (I read about nand support somewhere in web for this board)? And also add 1-2-3 wire support would be good. I noob but trying MN101EF31G MCU firmvare dumping MCU have a "Clock Synchronous Serial" port 2 or 3 wire adjustable, my board have a 2-wire type programming port, Bus pirate 2-wire is I2C based but I suspect this mcu Full or Half duplex SPI based (bus speed up to 5mhz datasheet). Some people tied MOSI and MISO together for half duplex SPI but one of the pin must be hi-z status other is working condition and other problem I can't understand "AAAh AAh 555h 55h" sequence in flash command table, datasheet says command address = 0x04000 + addr .Command data ? .Can you any advice for firmvare dumping for me or any source link? I am really helpless for this thing and I open couple of thread in forums but nobody help me. Thanks, sorry for my English.

bvernoux, 2015-04-23:
Such feature (NAND 8 or 16bits) Is planned in TODO list but not in top priority (as I'm the only one developer doing this project in my spare time) as there is lot of things to do before, if a developer is motivated (and have time & hardware to test it) I can even provide free HydraBus hardware for such task.

bvernoux, 2015-12-12:
I'm waiting hardware to test NAND/NOR Flash (mainly 48-Pin TSOP), tests will be done on:
* Hynix [HY27US08281A](http://www.hynix.com/datasheet/eng/nand/details/small_11_HY27US08281A.jsp) NandFlash 3.3V 16Mbyte
* Spansion [S29GL064A90TFIR4](http://www.cic.unb.br/~fbvidal/altera/Datasheets/Memory/flash/s29gl-a_00_a9_e.pdf) NandFlash 3.3V 8MByte (x8/x16)
* Spansion [S29AL016J70TFI020](https://www.spansion.com/Support/Datasheets/S29AL016J_00.pdf) NandFlash 3.3V 2MByte (2 M x 8-Bit/1 M x 16-Bit) 
* Samsung [K9F1G08U0C](http://datasheet.octopart.com/K9F1G08U0C-PCB000-Samsung-datasheet-8330961.pdf) NandFlash 3.3V 16Mbyte (x8) from http://www.waveshare.com/wiki/NandFlash_Board_(A)
* Samsung [S29GL128P](http://www.spansion.com/Support/Datasheets/S29GL-P_00.pdf) NORFlash 56TSOP (x16) (to test with [flashrom](http://flashrom.org/Flashrom) support) from http://www.waveshare.com/wiki/NorFlash_Board_(A)

relax79, 2016-02-05:
Hello , i am understand and thanks for your answer. Some people including me search this type cheap or diy project for usb flash and ssd data recovery(nand programmers are very expensive). These features would be nice if it were. I am so sorry for the delayed answerSorry for my English, thanks.

bvernoux, 2016-11-16:
NandFlash support is coming Read ID already work with my Samsung NandFlash (Samsung K9F1G08U0C NandFlash 3.3V 16Mbyte (x8) (TSOP48) from http://www.waveshare.com/wiki/NandFlash_Board_(A))
see https://github.com/bvernoux/hydrafw/tree/flash

Baldanos, 2016-12-15:
Made some changes in order to get this mode working.
Right now, it is possible to read/write 8x NANDs with the patches included in [this branch](https://github.com/bvernoux/hydrafw/tree/flash)

To test it, you can use [this modified version of DumpFlash](https://github.com/Baldanos/DumpFlash)

Demo : 
```
~/Projects/DumpFlash > python2 DumpFlash.py -i                                                                                                                                                   [master]
Into BBIO mode
Switching to flash mode
Setting chip enable
Full ID:	AD73AD73AD73
ID Length:	6
Name:		NAND 16MiB 3,3V 8-bit
ID:		0x73
Page size:	0x200
OOB size:	0x10
Page count:	0x8000
Size:		0x10
Erase size:	0x4000
Block count:	1024
Options:	0
Address cycle:	3
Bits per Cell:	4
Manufacturer:	Hynix
```

Baldanos, 2016-12-18:
NAND 8x mode has been added to hydraFW (https://github.com/bvernoux/hydrafw/commit/5247a8379c008bc5c481aa8afc3f5dc9e2c6a620)

Baldanos, 2018-04-08:
Since Hydraflash is now available for purchase (https://lab401.com/collections/hardware/products/hydraflash), we can consider this issue as closed.

closed 2018-04-08 by Baldanos