hydrabus/hydrafw #13: NFC scan doesn't work any longer

tuxscreen, 2015-05-06:
In the last releases I cant get nfc scan to work, with an older release (seeed original?) it worked, I updated to the commit 4.5.2015, got the SPI failures (you fixed that already), and now I got these nice sniffing, but scan is just dead. Am I missing something?

BR
tuxscreen

bvernoux, 2015-05-06:
It could be an issue with the new SPI speed set to 5.25MHz in order to have a better/faster emulation
Could you try the latest trunk and change
https://github.com/bvernoux/hydrafw/blob/master/hydranfc/hydranfc.c#L196
mode_con1.proto.dev_speed = 5; /* 5 250 000 Hz */
to
mode_con1.proto.dev_speed = 4; /* 2 620 000 Hz */
You can also try other value before it was set to 2 => 650 000 Hz

For Info SPI2 conf => mode_con1.proto.dev_speed
0 = 160000,
1 = 320000,
2 = 650000,
3 = 1310000,
4 = 2620000,
5 = 5250000,

Tell me if that fix the problem and which speed fix it.

tuxscreen, 2015-05-06:
Hi Benjamin,
I did a clean checkout of the current HEAD and rebuilt the binary from scratch. It works again.
I'm very sorry, it might be just my build Environment which got broken.
If I see something similar again, I'll try with different SPI Speeds

tuxscreen, 2015-05-06:
Addendum: I currently have four+ tags to test, and a Galaxy S3 to sniff.
The tags are one one Hand the ones you got with the TI TRF development kit, and I got some Mifare Ultralight tags, both are ISO14443-3A but different shape.

closed 2015-05-06 by tuxscreen

bvernoux, 2015-05-06:
Thanks for the feedback, I'm happy finally all is ok as I have tested last commit in trunk with Vicinity & Mifare One Tag without any issue.