hydrabus/hydrafw #21: USB sniffer capability

flabbergast, 2015-09-16:
An open source / HydraBus USB packet sniffing would be really cool! The commercial USB analyser solutions are way above (my) hobbyist budget; HydraBus + OTG cable would be highly appreciated.

I know you list this feature on your long-term todo/idea list, with "depending on feedback". So I thought I voice my support for this particular feature here. Feel free to close this if this is not the right place to ask :)

bvernoux, 2019-09-30:
For information see WIP https://github.com/0x8008135/hydrafw/tree/usb

bvernoux, 2022-09-22:
This feature have not provided any feedback from user's and there is no any contributor working on it too
If there is no any interest(feedback) at end of 2022 it will be closed

closed 2023-03-24 by bvernoux