hydrabus/hydrafw #23: Decoding sniffed NFC exchanges

bvernoux, 2015-10-27:
nickludlam commented on Feb 16

I've just got hold of the HydraBus + HydraNFC hardware, and I've sniffed my first NDEF record read + write.

With the text file, is there any way I can use a graphical tool to visualise the exchanges? I've looked at Wireshark-NFC, but that expects pcap format files. Would there be an easy way to convert the ascii sniffer output to pcap at all?

bvernoux, 2015-10-27:
bvernoux commented on Feb 16

Yes a pcap ng format will be a very nice feature.
I have planned that and need to add it in TODO also any help/suggestions are welcome.
The idea will be to save at end both format .txt and .pcapng with same name
The best will be to add that idea in hydrafw issue here https://github.com/bvernoux/hydrafw/issues

bvernoux, 2015-10-27:
stupidlinh commented 18 hours ago

I will try to write a piece of code to translate it to pcap.

bvernoux, 2015-10-27:
bvernoux commented 18 hours ago

Nico has done such work see https://github.com/NicoHub/Wireshark-RFID-dissector
I'm just waiting he release the firmware source code to merge it on hydrafw

AAsyunkin-se, 2019-09-29:
Hi Benjamin,

I imported/merged Nico's changes into my fork (no commits so far) - just to have a look.

While this looked promising initially, I am not too sure if it needs to be developed further: Wireshark LUA script for dissection is not context aware, i.e. you can't 'remember' the mode, the card is in, to interpret a message correctly.

This greatly decreases the value of decoding as ideally, say, for 14443-4 you'd want to work with PCB frames to see complete APDUs; determine 'past ATS' stage for it etc.

This, in turn, makes me think that a simple-ish Python script, which would be context-aware, would do a much better job, not only decoding all protocol levels, but providing an APDU export that can then be decoded using other tools (or in fact in this very script). But then such a script can work on existing text output as it is today. PCAP is not really needed.

Do you think it's still worth working on?

PS In addition to UART, do you think SPI2 can be used for 'eternal sniffing'? That would help recording on RPi or clones where SPI is reliable whereas UART has clock issues etc.

bvernoux, 2019-09-29:
It is an amazing project to do that with Python (Python 3.x as 2.x is dead now) 
You are welcome to work on that and I'm impatient to see the result (with a bonus like an HydraNFC v2)

About using SPI2 for 'eternal sniffing' as of today SPI2 is used to communicate with the NFC chipset to configure it for sniffer mode and then it is free and so it is possible to re-use it with RPi in order to stream the sniffed data for example (with RPi as SPI slave) as I suspect it is what you would like to do ?
An other simpler / better alternative will be to use USART1 (which is free) as SPI (using USART1_CK (PA8) / USART1_TX(PA9) / USART1_RX(PA10)) with minimal SW modifications in order to connect it to a RPi SPI Slave and stream data on this link (HydraBus USART1 SPI Mode => RPi SPI Slave)

We can also discuss about that in live on IRC Freenode channel #hydrabus (my nickname is bvernoux)

AAsyunkin-se, 2019-10-28:
First draft of a Python Script is available here - https://github.com/AAsyunkin-se/nfcdecoder

bvernoux, 2019-10-28:
Very nice work congratulations !!

bvernoux, 2020-01-29:
Implemented in https://github.com/hydrabus/hydrafw/commit/3c34dfd2f0cc2cb7cdf71b24267ce722490c24e0

closed 2020-01-29 by bvernoux