hydrabus/hydrafw #43: HydraNFC add SDM TX/DM1 RX mode to implement custom Reader/Writer

bvernoux, 2016-03-29:
Work in progress using TI [sloa214 "Using Special Direct Mode With the TRF7970A Feb2015"](http://www.ti.com/general/docs/litabsmultiplefilelist.tsp?literatureNumber=sloa214)
- Direct Mode Transceive => SDM TX then DM1 RX with management of Parity (if enabled)  & CRC (if enabled) 
- Separate mode SDM TX
- Separate mode DM1 RX

gibix, 2017-01-08:
There is some working progress code for this issue?

bvernoux, 2017-01-09:
It is always planned with lot of great features for 2017
For information the HydraNFC part will be separated (removed from main hydrafw) and moved to an other repository stay tuned

bvernoux, 2019-11-14:
This feature is cancelled for HydraNFC v1 but it will be integrated with HydraNFC v2 which use an other chipset much more powerful in all area and have also similar advanced mode for custom Emulation / Reader / Sniffer ...

bvernoux, 2020-01-29:
Closed
Feature cancelled for HydraNFC v1 but will be supported with HydraNFC v2 with different way to do low level emulation/Advanced Reader/Sniffer

closed 2020-01-29 by bvernoux