hydrabus/hydrafw #49: PulseView oscilloscope

mashu, 2016-04-24:
Hi,

Would it be possible to adapt HydraBus firmware such that it's recognized by http://sigrok.org/wiki/PulseView in ADC mode ?
I am looking for a cheap low speed 3.3V oscilloscope that can sample I2C bus running at 400kHz speed.
What matters is the shape of the peaks, simple diagnosis tool. Can HydraBus be used like that ?

Thanks

bvernoux, 2016-04-24:
Yes it can be clearly used like that as the ADC can run at up to 7MSPS (12bits data Dual Mode Interleaved) .
For information this feature is planned https://github.com/bvernoux/hydrafw/wiki/Task-List#hydrafw-todoidea => Basic Scope up to 3.3V and up to 7.2MSPS/12bits

bvernoux, 2019-11-14:
Any contribution is welcome about that feature thanks to say if you are working on it here

bvernoux, 2022-09-22:
This feature have not provided any feedback from user's and there is no any contributor working on it too
If there is no any interest(feedback) at end of 2022 it will be closed