hydrabus/hydrafw #56: Please set the release in the DFU releases

hughsie, 2016-09-01:
I noticed that the `hydrafw.dfu` file has no release set on the firmware file; this means that we can't notify the user if we try to downgrade the firmware:

    dfu-tool dump hydrafw.dfu 
    Loading hydrafw.dfu:
    vid:         0x0483
    pid:         0xdf11
    release:     0x0000 [0.0]
    format:      dfuse [0x0003]
    cipher:      none

when it probably should be something like:

    ...
    release:     0x0007 [0.7]
    ...

I guess the best way to do this would be to add an argument to dfu-convert.py in scripts and then replace the '0' argument in `<4H3sB` with the integer. Of course, you'd need to wire up the argument to dfu_convert_hydrafw.bat and the hydrafw-version.sh file, although you'd be more familiar with that than me.

If we do the LVFS thing this isn't *so* important (as the version is specified in the XML file) although it's best practice to set the release in any case.

bvernoux, 2016-09-02:
I need to check how to do that automatically has release version (embedded in firmware) are automatic and based on git informations

Baldanos, 2018-09-24:
Sent PR#84 to correct this

bvernoux, 2018-09-30:
I confirm it is fixed in new version 0.9 of hydrafw.dfu see https://github.com/hydrabus/hydrafw/releases/download/v0.9-beta/build_HydraFW_v0.9-beta-0-g17a50ac_HydraBus_HydraNFC.zip

dfu-tool dump build_HydraFW_v0.9-beta-0-g17a50ac_HydraBus_HydraNFC/build/hydrafw.dfu
Loading build_HydraFW_v0.9-beta-0-g17a50ac_HydraBus_HydraNFC/build/hydrafw.dfu:
vid:         0x0483
pid:         0xdf11
release:     0x0009 [0.9]
format:      dfuse [0x0003]
cipher:      none
= IMAGE 0 =
alt_setting: 0x00
name:        ST...
elements:    0x01
== ELEMENT 0 ==
address:     0x8000000
contents:    0x22a78

Baldanos, 2018-09-30:
Since it's part of the build process now, we can close this issue

closed 2018-09-30 by Baldanos