hydrabus/hydrafw #57: compilation errors gcc 4.8.4 ubuntu 14.04

iceman1001, 2016-09-22:
Minor compilation errors on a ubuntu 14.04 setup with gcc 4.8.4

http://pastebin.com/diKTF1MZ

iceman1001, 2016-09-22:
The DWORD seemed already to existed.  Changed it in code.
The last warning doesnt like converting uint8_t to uint32_t just like that. Added a helper function.
---
Got a final warning...

Linking build/hydrafw.elf
arm-none-eabi-gcc: error: nano.specs: No such file or directory

iceman1001, 2016-09-22:
Seemed to be the arm-gcc spoking around... but now I got new errors...

closed 2016-09-22 by iceman1001

bvernoux, 2016-09-22:
I confirm GCC ARM 4.8.x is totally buggy and shall be avoided (especially GNU ARM GCC 4.8 2014q3 which crash to build chibios with LTO option ...)
It is the same for GCC ARM Embedded 5.0/5-2016-q1 (build/link crash) see my bug report https://bugs.launchpad.net/gcc-arm-embedded/+bug/1570613

For information the build work fine with GNU ARM GCC 4.9 2015q3 (https://launchpad.net/gcc-arm-embedded/+milestone/4.9-2015-q3-update) and it will be the next step to use it by default and  to update the documentation, this new version improve lot of stuff (mainly for debug with GDB which support python ...)