hydrabus/hydrafw #63: Add support for SPI binary commands used by avrdude for buspirate AVR programming

david-sawatzke, 2016-11-09:
A sample:
http://pastebin.com/Jqz0jGKX
The non-supported command seem to be:
0x05 Write-then-read with no chip select with zero length (This seems like a bug)
0x01 Enter raw SPI mode , display version string
0x06 AVR Extended Commands 

These may not be all, as programming hasn't started.

bvernoux, 2016-11-09:
I confirm in actual version described here https://github.com/bvernoux/hydrafw/wiki/HydraFW-Binary-SPI-mode-guide
* 0x05 Write-then-read with no chip select with zero length (This seems like a bug) => Could you explain what is expected and what you obtain ?
* 0x01 Enter raw SPI mode , display version string => It is not supported but I think it can be easily added
* 0x06 AVR Extended Commands => It is not supported but I think it can be easily added

david-sawatzke, 2016-11-10:
A sample session with an BusPirate v4 (without a chip attached):
http://pastebin.com/9nyfcxCx

The bus pirate returns 0x01:
```
avrdude: buspirate_send_bin():
    0x05 0x00 0x00 0x00   0x00
avrdude: buspirate_recv_bin():
    0x01 
```

The Hydra bus doesn't return anything and seems to block:
```
avrdude: buspirate_send_bin():
        0x05 0x00 0x00 0x00   0x00
avrdude: ser_recv(): programmer is not responding
```

The BBIO_SPI_WRITE_READ_NCS command has to functions where zero-length could matter:
				`chnRead(con->sdu, tx_data, to_tx);`
and
```
bsp_spi_write_u8(proto->dev_num, tx_data,
				                 to_tx);
```
chnRead is symliked to streamRead, which is symlinked to `((ip)->vmt->read(ip, bp, n))`.
That seems to be the function:
`static size_t read(void *ip, uint8_t *bp, size_t n)`
which calls (hydrafw/chibios/os/hal/src/hal_buffers.c):
```
size_t ibqReadTimeout(input_buffers_queue_t *ibqp, uint8_t *bp,
                      size_t n, systime_t timeout)
```
which in turn calls
 `     memcpy(bp, ibqp->ptr, size); //size is 0
`
Which should work according to : https://stackoverflow.com/questions/3751797/can-i-call-memcpy-and-memmove-with-number-of-bytes-set-to-zero

bsp_spi_write_u8 calls HAL_SPI_Transmit wich handles Size == 0 as an error case and returns HAL_ERROR, which prompts bsp_spi_write_u8 to call spi_error, which in turn deintializes and reinitializes the SPI peripheral.
This may cause the non-response, maybe because of a timeout (but I'm not sure)

bvernoux, 2016-11-10:
Thanks for the details
I will try to implement SPI AVRDude ASAP
If you are interested for a patch proposal you are welcome like for any other contribution, just check 
https://github.com/bvernoux/hydrafw/wiki#how-to-help before to do any Pull requests (mainly coding rules)

Baldanos, 2016-11-10:
Hi !
Thanks for your submission.
I implemented fixes for this issue, however I don't have an AVR at hand.

Could you check if PR #64 works correctly with a real device ?

david-sawatzke, 2016-11-10:
There is another problem (I'm currently experimenting)
Bus Pirate:
```
Bulk SPI allows direct byte reads and writes. The Bus Pirate expects xxxx+1 data bytes. Up to 16 data bytes can be sent at once, each returns a byte read from the SPI bus during the write.

Note that 0000 indicates 1 byte because there's no reason to send 0. BP replies 0x01 to the bulk SPI command, and returns the value read from SPI after each data byte write. 
```
Hydra Bus:
`In this mode, the last 4 bits of the command define the number of bytes to write (from 1 to 16) (Command 0b00010000 will send 1 byte). The same number of bytes will be read and sent back to the user. Hydrabus will wait for the defined number of bytes, send a 0x01 (acknowledge) then the read bytes.`

AVRDude will wait for the HydraBus to return and time out. As AVRDude writes the bytes first (at least as far as I tested) and then reads them all, simply changing the order in which the 0x01 Byte is sen't seems to be sufficient
[0001-Change-ordering-of-confirm-Byte.patch.txt](https://github.com/bvernoux/hydrafw/files/584402/0001-Change-ordering-of-confirm-Byte.patch.txt)

[0001-Change-ordering-of-confirm-Byte.patch.txt](https://github.com/bvernoux/hydrafw/files/584397/0001-Change-ordering-of-confirm-Byte.patch.txt)

Baldanos, 2016-11-10:
You are correct, the return byte was not set at the right time. I updated PR #64 and now AVRDude returns this, whoch should contain the AVR response :

```
Attempting to initiate BusPirate binary mode...
avrdude: buspirate_send_bin():
        0x0a 0x0a
avrdude: buspirate_send_bin():
        0x00 0x00 0x00 0x00   0x00 0x00 0x00 0x00
        0x00 0x00 0x00 0x00   0x00 0x00 0x00 0x00
        0x00 0x00 0x00 0x00
avrdude: buspirate_recv_bin():
        0x42 0x42 0x49 0x4f   0x31
BusPirate binmode version: 1
avrdude: buspirate_send_bin():
        0x01
avrdude: buspirate_recv_bin():
        0x53 0x50 0x49 0x31
BusPirate SPI version: 1
avrdude: buspirate_send_bin():
        0x05 0x00 0x00 0x00   0x00
avrdude: buspirate_recv_bin():
        0x01
avrdude: Paged flash write enabled.
avrdude: buspirate_send_bin():
        0x49
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0x60
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0x8a
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0x06
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0x01
avrdude: buspirate_recv_bin():
        0x01 0x00 0x01
AVR Extended Commands version 1
avrdude: buspirate_send_bin():
        0x48
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0x13
avrdude: buspirate_recv_bin():
        0x01
avrdude: buspirate_send_bin():
        0xac 0x53 0x00 0x00
avrdude: buspirate_recv_bin():
        0xff 0xff 0xff 0xff
avrdude: initialization failed, rc=-2
         Double check connections and try again, or use -F to override
         this check.
```

david-sawatzke, 2016-11-10:
BBIO_SPI_CONFIG_PERIPH needs to support CS for avrdude to reset the avr
[0001-Implement-SPI-Select-Unselect.patch.txt](https://github.com/bvernoux/hydrafw/files/584600/0001-Implement-SPI-Select-Unselect.patch.txt)
The BBPIO_SPI_CONFIG bits don't mean the same for the HydraBus and the Bus Pirate:
`This command configures the SPI settings. Options and start-up defaults are the same as the user terminal SPI mode. w= pin output HiZ(0)/3.3v(1), x=CKP clock idle phase (low=0), y=CKE clock edge (active to idle=1), z=SMP sample time (middle=0). The Bus Pirate responds 0x01 on success. `
```
This allows to set the following parameters :

    x sets the polarity value
    y sets the clock phase
    z sets the SPI device (0=SPI1 or 1=SPI2)

```
This patch makes the timing look similar (not sure if it handles the other other cases correctly):
[0002-Make-polarity-setting-more-similar-to-BusPirate.patch.txt](https://github.com/bvernoux/hydrafw/files/584602/0002-Make-polarity-setting-more-similar-to-BusPirate.patch.txt)

This is the signal (both patches applied) with both the BusPirate and the HydraBus (to be used with sigrok) for:
[session.sr](https://github.com/bvernoux/hydrafw/files/584607/session.sr.txt)
Command:
`avrdude -c buspirate -P /dev/ttyACM1 -p attiny45 -x "spifreq=2" -vvvvv & avrdude -c buspirate -P /dev/ttyACM0 -p attiny45 -x "spifreq=0" -vvvvv`
The AVR still doesn't respond to the HydraBus and I'm not yet sure why.

Baldanos, 2016-11-11:
I added your patches (thanks BTW), and I made it working on a m32u4, but only with terminal commands in a local branch (https://github.com/Baldanos/hydrafw/tree/bbio). Are you able to test if you can at least make a `sig` command ?

```
> avrdude -c buspirate -P /dev/hydrabus -p m32u4 -t                                                                 [bbio]

Attempting to initiate BusPirate binary mode...
avrdude: Paged flash write enabled.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude> dump flash 256 256
>>> dump flash 256 256
0100  00 00 04 08 02 01 10 40  80 40 10 20 40 80 40 80  |.......@.@. @.@.|
0110  08 02 04 01 80 40 20 10  02 01 10 80 10 20 40 40  |.....@ ...... @@|
0120  04 04 04 04 04 03 04 05  02 02 02 02 04 03 02 02  |................|
0130  02 02 06 06 06 06 06 06  04 04 02 02 02 04 00 00  |................|
0140  00 00 25 00 28 00 2b 00  2e 00 31 00 00 00 00 00  |..%.(.+...1.....|
0150  24 00 27 00 2a 00 2d 00  30 00 08 0b 00 02 02 02  |$.'.*.-.0.. ....|
0160  01 00 09 04 00 00 01 02  02 00 00 05 24 00 10 01  |.. .........$...|
0170  05 24 01 01 01 04 24 02  06 05 24 06 00 01 07 05  |.$....$...$.....|
0180  81 03 10 00 40 09 04 01  00 02 0a 00 00 00 07 05  |....@ .... .....|
0190  02 02 40 00 00 07 05 83  02 40 00 00 00 c1 80 81  |..@......@......|
01a0  c1 12 01 00 02 02 00 00  40 41 23 37 80 00 01 01  |........@A#7....|
01b0  02 00 01 12 01 00 02 00  00 00 40 41 23 37 80 00  |..........@A#7..|
01c0  01 01 02 00 01 18 03 41  00 72 00 64 00 75 00 69  |.......A.r.d.u.i|
01d0  00 6e 00 6f 00 20 00 4c  00 4c 00 43 00 22 03 41  |.n.o. .L.L.C.".A|
01e0  00 72 00 64 00 75 00 69  00 6e 00 6f 00 20 00 4d  |.r.d.u.i.n.o. .M|
01f0  00 69 00 63 00 72 00 6f  00 20 00 20 00 20 00 04  |.i.c.r.o. . . ..|
```

david-sawatzke, 2016-11-12:
It seems like the frequency (320kHz) is to fast for the attiny45 with the default 1 MHz clock. When disabeling the clock divisor fuse, the communication works. I don't know how the speed of the SPI could be lowered. Maybe if SPI2 is used, it works? (That still won't be helpful when the AVR is set to internal 128kHz oscillator)

Baldanos, 2016-11-13:
The frequency selection on the SPI controller is done via a prescaler, so we can at most divide the base frequency by 8. It would require us to lower the clock on the STM32 to meet the 250kHz requirement, and we cannot do it properly at the moment.

Could you please test the PR #66 ? It includes your last patches and your code comments. If it's OK on your side, I'll push these changes and close the issue.

Thanks again for your contribution.

bvernoux, 2016-11-13:
For information see HydraBus STM32F4x5RGT Clock Configuration:
[image]

Like you see for your use case you could change directly APB1 or APB2 Prescaler to divide the SPI clock ...

david-sawatzke, 2016-11-13:
The Hydrabus doesn't program the chip (When reading the memory back out it's all 0xff)
This even happens in terminal mode
It seems that the Hydrabus changes the reset line state more often than the buspirate, but I didn't yet check if that's be the reason and why it happens
[hydraterm.txt](https://github.com/bvernoux/hydrafw/files/587834/hydraterm.txt)
[hydrabus.txt](https://github.com/bvernoux/hydrafw/files/587846/hydrabus.txt)
[buspirate.txt](https://github.com/bvernoux/hydrafw/files/587847/buspirate.txt)
[hydrafw.sr.txt](https://github.com/bvernoux/hydrafw/files/587832/hydrafw.sr.txt)
[buspirate.sr.txt](https://github.com/bvernoux/hydrafw/files/587831/buspirate.sr.txt)

david-sawatzke, 2016-11-13:
And it would be really desirable to support an AVR with a 1MHz clock, as most of the most popular ones (as the Atmega328 and the attiny45) have that as their default frequency. Maybe simply defaulting to SPI2 would be the best solution

Baldanos, 2016-11-14:
Hi !

I updated the PR #66 with some bugfixes. It now works correctly with avrdude, both in reading and writing to flash on my m32u.
I also made a change in the BBIO_SPI_CONFIG subcommand to select SPI2 instead of SPI1. Tell me if it works for your device as well.

david-sawatzke, 2016-11-14:
Seems to work fine

Baldanos, 2016-11-14:
I did some more tests, and it seems to be working quite well.
I merged this PR and I'll close this issue.

Thanks again for your feedback and tests

closed 2016-11-14 by Baldanos