hydrabus/hydrafw #71: BDM/Coldfire support

csholmq, 2017-01-17:
I have come across a Freescale chip that uses BDM, instead of JTAG, which I wasn't familiar with since before. I noticed OpenOCD has it on its todo-list but has yet to produce anything.

Would it make sense to have this supported by Hydrabus?

Baldanos, 2017-01-17:
Hi !

The BDM communication could be easily added to Hydrabus (according to http://www.nxp.com/files-static/training_pdf/27642_HCS08_BDM_ICE_WBT.pdf). However, I couldn't find any opensource tool that uses a serial interface for the BDM probe. It would therefore be difficult to interface it with GDB right away.
But still, creating a flash/dump script using BBIO commands in this mode should be doable.

Would you be able to help on testing this ?

csholmq, 2017-01-25:
It appears that the chip I am looking at is locked, so at the moment it would prove difficult as I don't have a board to test on. I will see if I can get my hands on one for debugging purposes.

csholmq, 2018-01-29:
About one year later I am better prepared for testing this :) I am acquiring a test board so that I will soon have a setup.

How can I get started on reading up on this?

csholmq, 2018-02-16:
@Baldanos So I've started writing up a script with BBIO commands. But with BDM being asynchronous between host and target clock, timing is everything.

E.g for transmitting a logical 0/1, if the target is running at 8 MHz, I need to initiate communication with a negative flank and then set the BKGD HIGH/LOW within 8 cycles (=1 us). Is this possible with BBIO commands?

All according to:
[image]

Baldanos, 2018-02-21:
@csholmq This protocol looks a bit similar to 1-wire, except the clock is not fixed. There is currently no way to use this protocol with Hydrabus, but it doesn't look too difficult to implement. 

I think the best way to start would be to implement the SYNC command, and use the calculated clock to program a timer. That timer can then be used to implement a read_bit and write_bit primitive, then write_byte and read_byte.

Take a look at src/hydrabus/hydrabus_mode_onewire.c for an example of that kind of protocol.

I'd be glad to help if you want to implement this on Hydrabus.

bvernoux, 2019-11-30:
Any news on this subject as so far it seems no anyone have the hardware (or hardware and time to work on it) to do test and implement it ?

bvernoux, 2020-09-12:
Issue closed as there is no new to support that specific Target

closed 2020-09-12 by bvernoux