hydrabus/hydrafw #76: HydraBus hangs when using flashrom

pedrib, 2017-08-30:
HydraBus is supposed to be binary compatible with the Bus Pirate, so flashrom should work out of the box (flashrom has a bus pirate mode).

However, when trying to ID a chip using flashrom (the most basic operation before a read or write), flashrom just hangs:
```
tools > sudo flashrom -p buspirate_spi:dev=/dev/ttyACM0,spispeed=8M
flashrom v0.9.9-r1954 on Linux 4.4.39-grsec+ (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
<HANGS FOREVER>
```

pedrib, 2017-08-30:
Different speeds don't seem to make a difference.

Baldanos, 2017-09-12:
Changes have to be made both in HydraFW and FlashROM.

 - Support for Hydrabus has been added to hydrafw [8488be1e5231ca5e55c4ba193aebaf1b29abcfd1](https://github.com/hydrabus/hydrafw/commit/8488be1e5231ca5e55c4ba193aebaf1b29abcfd1)
 - A flashrom pull request is ongoing to support it upstream : (https://github.com/flashrom/flashrom/pull/21)

You can test it on (https://github.com/Baldanos/flashrom)

pedrib, 2017-09-12:
Thanks! 
I'm travelling at the moment, give me 10 days and I'll get back to you.

dr0pd34d, 2017-12-20:
Hey there @Baldanos do you have a HydraBus?
Can you recommend it for firmware dumping and JTAG?
Does you fork work for reading flash memory correctly?

thank you :)

afiskon, 2018-07-04:
Unfortunately the problem still exists:

```
eax@e733:~$ flashrom -p buspirate_spi:dev=/dev/ttyACM0,spispeed=1M --read flash.dump -VV
flashrom v1.0 on Linux 4.17.3-1-ARCH (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.5.6, GCC 7.2.1 20171224, little endian
Command line (5 args): flashrom -p buspirate_spi:dev=/dev/ttyACM0,spispeed=1M --read flash.dump -VV
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing buspirate_spi programmer
(HANGS FOREVER)
```

BusPirate dumps the same SPI flash (AT45DB041D) with the same flashrom without any problems. The chip is connected properly, I can read device id on spi1:
```
spi1> [ 0x9F r:5 ]
/CS ENABLED
WRITE: 0x9F
READ: 0x1F 0x24 0x00 0x00 0x00 
/CS DISABLED
```

Here is an output of strace: https://afiskon.ru/s/29/b16555a066_strace.txt

I'm using relatively new HydraFW compiled from cb077a7 (Thu May 24 08:21:25 2018 +0200).

Baldanos, 2018-07-07:
Hi !

The official revision of Flashrom still does not includes our patch to support Hydrabus (see https://github.com/flashrom/flashrom/pull/21)

I have a (outdated) fork here if you want to test : https://github.com/Baldanos/flashrom

Tell me if it works for you as well, so we can ask the PR to be merged.

Baldanos, 2018-07-07:
Thanks for the feedback.

Unfortunately, I don't have EEPROMs at hand to test, could you try to connect your EEPROM to SPI2 ?
```
CS:   PC1 (SW)
SCK:  PB10
MISO: PC2
MOSI: PC3
```
Long story short, we had to use SPI2 by default in binary SPI instead of SPI1 because of AVR programming uses very low speeds. See the Wiki for this : https://github.com/hydrabus/hydrafw/wiki/HydraFW-Binary-SPI-mode-guide#configure-spi-port-0b10000xyz

If that doesn't work, could you please run the same commands with the `-VVV` switch to flashrom and paste the results here ? I'll see if I can reproduce this next week.

afiskon, 2018-07-08:
Hello @Baldanos,

If I connect the chip to SPI2 flashrom (the forked one) recognizes the chip however it can't dump it:

```
$ ./flashrom -p buspirate_spi:dev=/dev/ttyACM0 --read flash.dump
flashrom v0.9.9-55-g7bb811f-dirty on Linux 4.17.3-1-ARCH (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Atmel flash chip "AT45DB041D" (528 kB, SPI) on buspirate_spi.
Reading flash... spi_read_at45db: error sending read command!
Read operation failed!
FAILED.
```

Here is a full -VVV log:

```
$ ./flashrom -p buspirate_spi:dev=/dev/ttyACM0 --read flash.dump -VVV
flashrom v0.9.9-55-g7bb811f-dirty on Linux 4.17.3-1-ARCH (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.5.6, GCC 8.1.1 20180531, little endian
Command line (5 args): ./flashrom -p buspirate_spi:dev=/dev/ttyACM0 --read flash.dump -VVV
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing buspirate_spi programmer
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 0, read 4 , receiving 0x42 0x42 0x49 0x4f
buspirate_sendrecv: write 1, read 6 Sending 0x0f, receiving 0x31 0x48 0x79 0x64 0x72 0x61
Detected Hydrabus hardware
SPI speed is 21MHz
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 1, read 0 Sending 0x00
[... 1231 lines cut ...]
RDID returned 0x1f 0x24 0x00. probe_spi_rdid_generic: id1 0x1f, id2 0x2400
Probing for Generic unknown SPI chip (RDID), 0 kB: buspirate_sendrecv: write 6, read 4 Sending 0x04 0x00 0x01 0x00 0x03 0x9f, receiving 0x01 0x1f 0x24 0x00
RDID returned 0x1f 0x24 0x00. probe_spi_rdid_generic: id1 0x1f, id2 0x2400
Probing for Generic unknown SPI chip (REMS), 0 kB: buspirate_sendrecv: write 9, read 3 Sending 0x04 0x00 0x04 0x00 0x02 0x90 0x00 0x00 0x00, receiving 0x01 0x00 0x00
REMS returned 0x00 0x00. probe_spi_rems: id1 0x0, id2 0x0
Found Atmel flash chip "AT45DB041D" (528 kB, SPI).
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
programmer_map_flash_region: mapping AT45DB041D from 0x00000000fff7c000 to 0x0000000000000000
buspirate_sendrecv: write 9, read 1 Sending 0x04 0x00 0x04 0x00 0x00 0x3d 0x2a 0x7f 0x9a, receiving 0x01
buspirate_sendrecv: write 6, read 2 Sending 0x04 0x00 0x01 0x00 0x01 0xd7, receiving 0x01 0x9c
Status register: 0x9c.
Reading flash... at45db_convert_addr: addr=0x0, page_size=264, page_bits=9 -> at45db_addr=0x0
spi_read_at45db: error sending read command!
Read operation failed!
FAILED.
programmer_unmap_flash_region: unmapped 0x0000000000000000
buspirate_sendrecv: write 1, read 0 Sending 0x00
buspirate_sendrecv: write 0, read 4 , receiving 0x42 0x42 0x49 0x4f
buspirate_sendrecv: write 0, read 1 , receiving 0x31
Raw bitbang mode version 1
buspirate_sendrecv: write 1, read 0 Sending 0x0f
Shutdown completed.
```

Baldanos, 2018-08-06:
Hello @afiskon !

I found a bug in my flashrom patch. It should be fixed now.
Can you please test again ?

```
> ./flashrom -p buspirate_spi:dev=/dev/ttyACM0 --read flash.dump                                [staging]
flashrom p1.0-26-g524bbe3-dirty on Linux 4.17.11-arch1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found GigaDevice flash chip "GD25Q32(B)" (4096 kB, SPI) on buspirate_spi.
Reading flash... done.
```

afiskon, 2018-08-07:
Hello @Baldanos,

I tested updated flashrom on AT45DB041D chip connected to SPI2 and it works:

```
$ ./flashrom -p buspirate_spi:dev=/dev/ttyACM0 --read flash.dump
flashrom v0.9.9-96-g6cbe42f-dirty on Linux 4.17.10-1-ARCH (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Atmel flash chip "AT45DB041D" (528 kB, SPI) on buspirate_spi.
Reading flash... done.

$ hexdump -C flash.dump | less
...
00025830  54 68 69 73 20 69 73 20  61 20 74 65 73 74 20 6d  |This is a test m|
00025840  65 73 73 61 67 65 00 ff  77 c8 07 f8 ba fe 35 d0  |essage..w.....5.|
...
```

Thank you a lot!

Baldanos, 2018-08-08:
@afiskon Great news !

FYI, I ported the patch to the current flashrom revision here : https://github.com/Baldanos/flashrom/tree/master

I hope the PR will get merged soon.

closed 2020-08-28 by Baldanos

Baldanos, 2020-08-28:
Since the PR (https://github.com/flashrom/flashrom/pull/21) is open for nearly three years now, I took some time to reimplement the [Serprog protocol](https://github.com/flashrom/flashrom/blob/master/Documentation/serprog-protocol.txt) in HydraFW. 

This works on any recent flashrom : 

```
$ flashrom -V --programmer serprog:dev=/dev/hydrabus
flashrom v1.2 on Linux 5.8.3-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with libpci 3.6.4, GCC 9.3.0, little endian
Command line (3 args): flashrom -V --programmer serprog:dev=/dev/hydrabus
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Initializing serprog programmer
No baudrate specified, using the hardware's defaults.
serprog: connected - attempting to synchronize
..
serprog: Synchronized
serprog: Interface version ok.
serprog: Bus support: parallel=off, LPC=off, FWH=off, SPI=on
serprog: Maximum write-n length is 4096
serprog: Maximum read-n length is 4096
serprog: Programmer name is "Hydrabus"
serprog: Serial buffer size is 16
serprog: Output drivers enabled
The following protocols are supported: SPI.
[...]
```

bvernoux, 2020-08-29:
Thanks for the commit https://github.com/hydrabus/hydrafw/commit/c10868f345027142b0bcea3c76b8ebd8c27d45c0
Which implement flashrom serprog

commit c10868f3 (2020-08-28): Add serprog protocol

Serprog is fully supported by flashrom. This fixes #76 (...)