hydrabus/hydrafw #79: HydraNFC add 1443A parity bits to sniffer traces

zhovner, 2018-01-11:
Proxmark3 can sniff parity bits for each byte. This handy for some cryptanalysis.  

Example of proxmark3 sniffer dump with parity bits marks.

```
Parity bit 1 indicates as "!" and parity bit 0 is empty.  

 +    112:  4 : TAG 5a! 92  0d! 85! 						
 +   6946:  8 :     98! d7  6b! 77  d6  c6  e8  70			
 +     64:  4 : TAG ca  7e! 0b! 63!						
 + 670868:  4 :     3e! 70  9c! 8a 
 +    112:  4 : TAG 36! 41  24! 79    
 +   9505:  8 :     1b! 8c  3a! 48! 83  5a  4a! 27
```

It will be nice to see this is hydrafw.

bvernoux, 2018-01-14:
The parity is already implemented with hydranfc_sniff_14443A_bin mode see https://github.com/hydrabus/hydrafw/blob/master/src/hydranfc/hydranfc_cmd_sniff.c#L990
For an example see https://github.com/hydrabus/hydratool/releases especially sniff bin frame-time parity_mifare_classic.bin and open it with hydratool of course you can also customize the display if you want ! ou space instead of 01 00 it is very easy to do that in hydratool as full source code is available

zhovner, 2018-01-14:
Cool, thank you! 
But I like standalone sniffer mode and wan't to see parity bits in this mode too. There is no technical limitations to do this?

Maybe I would try to implement it myself if my poor C-programming skills will be enough.

bvernoux, 2018-01-14:
All is done here in function hydranfc_sniff_14443A:
* https://github.com/hydrabus/hydrafw/blob/f0f2d6a30835362fa0fff218b6d9140c530b8ebd/src/hydranfc/hydranfc_cmd_sniff.c#L910
  * Just before this line you just need to retrieve the parity bit value with **miller_modified_106kb[ds_data]** then you can write it with sniff_write_Parity_PM3() to write "  " or "! " depending on the parity bit value

* https://github.com/hydrabus/hydrafw/blob/f0f2d6a30835362fa0fff218b6d9140c530b8ebd/src/hydranfc/hydranfc_cmd_sniff.c#L924
  * Just before this line you just need to retrieve the parity bit value with **manchester_106kb[ds_data]**  then you can write it with sniff_write_Parity_PM3() to write "  " or "! " depending on the parity bit value

You shall create your own sniff_write_Parity code to convert from parity bit value 0 or 1 to "  " or "! "
```
void sniff_write_Parity_PM3(uint8_t parity)
{
	uint32_t i;
	i = g_sbuf_idx;
	if(parity == 0)
		g_sbuf[i+0] = ' ';
	else
		g_sbuf[i+0] = '!';
	g_sbuf[i+1] = ' ';
	g_sbuf_idx +=2;
}
```

I let you test it yourself but like you can see it is very easy to add that

zhovner, 2018-01-16:
Seems that it works. https://github.com/zhovner/hydrafw/commit/19ae377a08d1259ba992148dd5091d9c08d57a99
Thank you for your advices.

That's what output I get now: 
```
3fdd1638	RDR	26  
3feb8254	RDR	52  
3febebaf	TAG	44! 00! 
3feee599	RDR	93! 70  88! 04  33! 18! a7  2b! 98  
3ff11957	TAG	04  da  17! 
3ff384cc	RDR	95! 20  
3ff4324a	U80	ff! ff! ff! ff! ff! 
3ff940f3	Uff	95! 70  ea  70  32  80  28! ca! ae  
3ffb74b3	TAG	08  b6  dd! 
412d4ccb	RDR	60! 1c  18! a1  
412f0ac6	TAG	02  ef  b4! 13  
4130341a	RDR	3c  ed! dd  97  39! 9c  98  19  
4132302d	TAG	3d! 9a! 74  02  
42341844	RDR	e7  29  95! ff  
4235c3c1	TAG	ed! 60! 8f! 2b  
4236ed0a	RDR	03! 40  88! c3! 0a  7e! e8  00! 
4238e910	TAG	01  d8! 9a! 9c! 
430dc8b2	RDR	57  1d  a6  56  
430f239c	TAG	a1! f8! 58  df  52  7c  48! af! 2f  dd  b1  0c! 46  d1! 50  45  a6  23  
```

If someone interesting to test it with me, here is the [firmware build](https://github.com/zhovner/hydrafw/releases/download/v0.8-beta/build_HydraFW-2e933dc_parityPM3-standalone-patch.zip).  
I summon @J-Run in this thread.

bvernoux, 2018-10-17:
A pull request is welcome if you confirm all work fine

zhovner, 2018-10-23:
Guess I misunderstood how proxmark3 parity bits works.
It doesn't just show the parity bit but also calculates parity from data and shows if it's wrong.

Here are the explanations:  
http://www.proxmark.org/forum/viewtopic.php?id=173
http://www.proxmark.org/forum/viewtopic.php?id=223

We also can calculate the wrong parity manually but I think it would be more useful to see wrong parity like proxmark shows.

bvernoux, 2018-10-24:
Yes maybe the best is to compute it like on proxmark3 with clear explanation about that in algorithm

bvernoux, 2020-01-29:
No update on this topic
Issue closed (to be reopened if someone want to implement it)

closed 2020-01-29 by bvernoux