hydrabus/hydrafw #82: Sigrok / PulseView compatibility seems to be broken

afiskon, 2018-07-01:
Hello,

I tried to use HydraFW with Sigrok according to the wiki article ( https://github.com/hydrabus/hydrafw/wiki/HydraFW-SUMP-guide ). However after entering SUMP mode PulseView doesn't find the device (screenshot: https://afiskon.ru/s/33/65d517b933_temp.png ), nor 'sigrok-cli --scan` does:

```
$ sigrok-cli --scan
The following devices were found:
demo - Demo device with 12 channels: D0 D1 D2 D3 D4 D5 D6 D7 A0 A1 A2 A3
```

I'm using HydraFW compiled from source, commit cb077a71 (Thu May 24 08:21:25 2018 +0200). Here are versions of PulseView and sigrok-cli, they are relatively new as well:

```
$ pulseview --version
PulseView 0.5.0-git-1931b5f
$ sigrok-cli --version
sigrok-cli 0.8.0-git-a30c837
```

I didn't investigate the problem deeply, maybe this is a Sigrok's bug. I will report this issue to sigrok-devel@ mailing list as well. From what I know BusPirate compatibility is broken in Sigrok as well https://sigrok.org/bugzilla/show_bug.cgi?id=1136

If anyone can reproduce this issue I suggest to edit the wiki page accordingly.

Baldanos, 2018-07-07:
Hi ! 

Thanks for the report !

Using the last revision from master, I tested it with sigrok 0.7 :
```
$ sigrok-cli --version                                                                                                                                       
sigrok-cli 0.7.0
$ sigrok-cli --driver=ols:conn=/dev/ttyACM0 --scan                                                                                                           
The following devices were found:
ols - HydraBus with 16 channels: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15
```
Maybe there is something different with sigrok 0.8. Could you please try again using the following command and send me the result ?
`sigrok-cli --driver=ols:conn=/dev/ttyACM0 --scan -l 5`

Also, since https://github.com/hydrabus/hydrafw/commit/ca90ec2a30b7e80f157d30756444a9b279ad2f90, hydrabus can be detected without the need to enter the `sump` command.

afiskon, 2018-07-07:
My bad, it actually works. I realized that I should choose "Serial Port" instead of "USB" in PulseView's dialog: https://afiskon.ru/s/92/86885bbfdb_temp.png

closed 2018-07-07 by afiskon