hydrabus/hydrafw #91: Driver installation WIN7 64

Tanergol, 2019-02-12:
hi,

after setting hydrabus in DFU mode (holding UBTN)

I am trying to install the WIN7 x64 driver from STMicroelectronics STSW-STM32080.

driver : STM Device in DFU Mode 
I always get : 
the device cannot start. error  code 10 

Any idea how to install the driver ?

bvernoux, 2019-02-13:
1) Did you have tried latest version STSW-STM32080 3.0.6 from https://www.st.com/en/development-tools/stsw-stm32080.html ?
Note it requires you create an account in order to download it
2) It could be an issue with the USB port used (especially on some USB 3.x port on some PC) could you try using an external USB 2.0 HS Hub ?
See details here https://github.com/hydrabus/hydrafw/wiki/Getting-Started-with-HydraBus-flash-and-use-hydrafw-on-windows#3-connect-microusb-cable-from-pc-to-hydrabus-usb1-board-now-windows-shall-detect-a-new-device
Extract of interesting part
Note: If you have error to update the firmware, change USB port or try with an External USB 2.0 Hub as there is sometimes problems with some USB2.0/USB3.0 ports, then retry this step.

Tanergol, 2019-02-13:
thanks for the update I did the flashing on my Linux machine. It worked just fine.

closed 2019-02-13 by Tanergol