hydrabus/hydrafw #93: Error "Frame size error resync data" using HydraFW v0.9 Beta

bvernoux, 2019-03-31:
The issue seems to be a bug to configure UART on latest HydraFW v0.9 (as it does not affect HydraFW v0.8) For more details see also https://github.com/hydrabus/hydratool/issues/2

I have done some fix especially one which should fix this issue (added missing explicit initialisation of bus_mode added in latest refactor => mode_conf.config.uart.bus_mode = BSP_UART_MODE_UART)
hydrabus/hydrafw@5204114
Could you rebuild the firmware with HydraFW master and tell me if that fix the issue

Note: This issue shall be part of HydraFW issue (even if I have fixed minor things at same time in this tool)

drspangle, 2019-04-17:
I seemed to have this error in HydraFW v0.8 and v0.9. In both firmware versions I consistently got this error along with some undecipherable bytes. The output was not consistent with the output from the `nfc sniff` command. That output was also garbled, and I'm not sure how to interpret it. I was using a [Sparkfun FTDI interface](https://www.sparkfun.com/products/9716), could this have been part of the problem? We used very short wires to connect to the pin header on the hydrabus with dupont connectors.

bvernoux, 2019-04-17:
SparkFun FTDI is a low cost chip LS/FS USB2.0 FT232RL which does not support required speed 8.4Mbauds but only something like 3Mbauds and so it is mandatory to use C32HM-DDHSL-0 (FT232H / FT2232H )

drspangle, 2019-04-17:
Alright, I'll revisit this once I have the correct FTDI interface.

bvernoux, 2019-05-13:
I confirm this issue is fixed with latest hydrafw master https://github.com/hydrabus/hydrafw/commit/55e5dbe39052edf71386fd4f24772f26c4ba2605
See archive build_HydraFW_v0.9-beta-65-g55e5dbe_HydraBus_HydraNFC.zip of the firmware pre-built
[build_HydraFW_v0.9-beta-65-g55e5dbe_HydraBus_HydraNFC.zip](https://github.com/hydrabus/hydrafw/files/3174246/build_HydraFW_v0.9-beta-65-g55e5dbe_HydraBus_HydraNFC.zip)

closed 2019-05-15 by bvernoux