hydrabus/hydrafw #94: Cannot set smartcard mode

thierryfischer, 2019-04-01:
Hello,

When trying to play the  [smartcard_bbio.py](../blob/master/contrib/bbio_smartcard/smartcard_bbio.py) script, it fails with the error "Cannot set smartcard mode, try again or reset hydrabus."
I tried to reset it, but all the time the response is 'BBIO', it does not switch to 'CRD1' as expected in the script.

I have pulled the last version of HydraFW and correctly build and flash it according to the user guide, indeed with the firmware I am able to able to get ATR.

Please find below details about the system information:

```
> show system
HydraFW (HydraBus) v0.9-beta-49-g5a3a20a-dirty 2019-03-23
sysTime: 0x000157a3.
cyclecounter: 0xf6a12f45 cycles.
cyclecounter64: 0x00000000f6a12f54 cycles.
10ms delay: 1680027 cycles.

MCU Info
DBGMCU_IDCODE:0x100F6413
CPUID:        0x410FC241
Flash UID:    0x30001A 0x31374707 0x33373331
Flash Size:   1024KB

Kernel:       ChibiOS 4.0.4
Compiler:     GCC 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
Architecture: ARMv7E-M
Core Variant: Cortex-M4F
Port Info:    Advanced kernel mode
Platform:     STM32F4x5 High Performance with DSP and FPU
Board:        HydraBus 1.0
Build time:   Apr  1 2019 - 11:14:50
```

Did anyone else experienced the same or manage to run that script correctly ?

Best regards,

bvernoux, 2019-04-01:
Could you test using latest master as there is a fix related to BBIO mode
https://github.com/hydrabus/hydrafw/commit/4995ea5ef4423df03318fdd1410aed138adba0e7

thierryfischer, 2019-04-02:
Thanks, but the error is still present with that fix.
```
> show system
HydraFW (HydraBus) v0.9-beta-50-g4995ea5-dirty 2019-03-29
...
```
I am running it on a windows host.

Baldanos, 2019-04-02:
Hi !

The readme says this script is made for Python 3.2+

I tested the following : 
Python 2.7.16 => Same error message as you
Python 3.7.3 => Works fine

Can you please verify that you are using a correct Python version ?

thierryfischer, 2019-04-02:
Thank you @Baldanos it works:
```
$ Python36-32/python bbio_smartcard/smartcard_bbio.py
Parsing ATR:
Convention direct: 0x3b
T0:  0x65
TB1: 0x0
TC1: 0x0, extra guard time integer N=0
TA2: absent, card in negotiable mode.
Historical bytes: 2063cbaba0
Traceback (most recent call last):
  File "bbio_smartcard/smartcard_bbio.py", line 280, in <module>
    convention = smartcard_parse_atr()
  File "bbio_smartcard/smartcard_bbio.py", line 253, in smartcard_parse_atr
    if protocol != 0:
UnboundLocalError: local variable 'protocol' referenced before assignment
```
Do you know where the incompatibility comes from in Python2.7 ? Byte encoding ?

Regards,

Baldanos, 2019-04-02:
Looks like python2 does not do the same with the `bytes()` function. On line 71 of the script you have 

```
hydrabus.write(bytes([BBIO_SMARTCARD]))
```

Python2 : 
```
>>> bytes([0b00001011])
'[11]'
```

Python3 (correct) :
```
>>> bytes([0b00001011])
b'\x0b'
```

I'll let you close the issue if it work for you.

Thanks for reporting !

closed 2019-04-03 by thierryfischer