hydrabus/hydrafw #98: HydraFW jtag brute scan reboot target device

ghecko, 2019-06-14:
With the latest firmware compiled from the Master Branch, the brute bypass & idcode scan reboot the target device.

bvernoux, 2019-06-19:
Could you explain why do you closed this issue (I suspect it was not an issue with HydraFW but it will be better to add an explanation) ?

ghecko, 2019-06-25:
Yes, when I used brute command on Hydrabus the target device reboot on each try. I can not reproduce this specific issue on another target device; this is why I close the issue.

bvernoux, 2019-06-27:
Thanks for the details if you could provide details on the target it could be interesting for other users ...

ghecko, 2019-07-05:
Yes, the target is a 4G modem Huawei B525s-23a (processor Hi6932M2GFCV100). If someone find the datasheet I want it !

bvernoux, 2019-07-06:
Thanks for the feedback but unfortunately no public doc exist (today) on this CPU/4G Modem especially related to JTAG
This issue is closed and you can reopen it if you have a datasheet or you have found a bug in JTAG brute scan

closed 2019-07-06 by bvernoux