hydrabus/hydrafw #100: UART Binary mode failed to change baudrate

ghecko, 2019-07-05:
I'm trying to implement a baurate.py like script for using it with hydrabus.
When I'm trying to change the baurate speed, I first stop echo UART UX, change the baudrate value and then activate again echo UART RX. On the second iteration, I'm not able to stop echo UART RX (receive 0xff).

My code :

```
import serial
import time

serial = serial.Serial('/dev/ttyACM0', 115200, timeout=5)

VOWELS = ["a", "A", "e", "E", "i", "I", "o", "O", "u", "U"]
WHITESPACE = [" ", "\t", "\r", "\n"]
PUNCTUATION = [".", ",", ":", ";", "?", "!"]
BAUDRATES = [{"dec": 9600, "hex": b'\x64'}, {"dec": 19200, "hex": b'\x65'}, {"dec": 38400, "hex": b'\x66'},
             {"dec": 57600, "hex": b'\x67'}, {"dec": 115200, "hex": b'\x6a'}]

def gen_char_list():
    c = ' '
    valid_characters = []
    while c <= '~':
        valid_characters.append(c)
        c = chr(ord(c) + 1)

    for c in WHITESPACE:
        if c not in valid_characters:
            valid_characters.append(c)
    return valid_characters

def set_baudrate(baudrate):
    # Deactivate echo UART RX
    print("Stop echo UART RX")
    serial.write(b'\x03')
    val = serial.read(1)
    print(val)
    if b'\x01' not in val:
        print("Cannot stop echo UART RX mode")
        return False

    # Change baudrate speed
    serial.write(baudrate["hex"])
    ret = serial.read(1)
    print("set baudrate ret={}".format(ret))
    if b'\x01' not in ret:
        print("Cannot set Baudrate")
        return False
    print("Selected baudrate: {}".format(baudrate["dec"]))

    # Activate echo UART RX Mode
    print("Start echo UART RX")
    serial.write(b'\x02')
    if b'\x01' not in serial.read(1):
        print("Cannot start echo UART RX mode")
        return False

    return True

def baudrate_detect():
    count = 0
    whitespace = 0
    punctuation = 0
    vowels = 0
    found = 0
    threshold = 25
    valid_characters = gen_char_list()

    print("Starting baurate detection, turn on your serial device now")
    print("Press Ctrl+C to cancel")

    for baudrate in BAUDRATES:
        if found == 0:
            if set_baudrate(baudrate):
                while True:
                    tmp = serial.read(1)
                    try:
                        byte = tmp.decode('utf-8')
                    except UnicodeDecodeError:
                        byte = tmp
                    print("byte={}".format(byte))
                    if byte:
                        if byte in valid_characters:
                            if byte in WHITESPACE:
                                whitespace += 1
                            elif byte in PUNCTUATION:
                                punctuation += 1
                            elif byte in VOWELS:
                                vowels += 1
                            count += 1
                        else:
                            whitespace = 0
                            punctuation = 0
                            vowels = 0
                            count = 0
                            break
                        if count >= threshold and whitespace > 0 and punctuation > 0 and vowels > 0:
                            found = 1
                            break
                    else:
                        break
            else:
                break
        else:
            break

def init_hb():
    print("Switching to BBIO mode")
    for i in range(20):
        serial.write(b'\x00')
    if "BBIO1".encode('utf-8') not in serial.read(5):
        print("Could not get into bbIO mode")
        quit()

    print("Switching to UART mode")
    serial.write(b'\x03')
    if "ART1".encode('utf-8') not in serial.read(4):
        print("Cannot set UART mode")
        quit()

if __name__ == '__main__':
    init_hb()
    baudrate_detect()
```

bvernoux, 2019-07-06:
Thanks for the bug report I think I have found the root cause of issue is in code  https://github.com/hydrabus/hydrafw/blob/master/src/hydrabus/hydrabus_bbio_uart.c#L43
As it does not contains a check like chThdShouldTerminateX() to exit from thread in a clean way and so I suspect this task is frozen and cannot be destroyed ...
I will test a fix and push it as soon as possible

bvernoux, 2019-07-06:
I have done a fix and tested it with success with your python script
I'm waiting your feedback

ghecko, 2019-07-06:
Thank you for your rapidity for the fix !
If I have no signal on hydrabus RX (no target device) the fix work. But if a device transmits to hydrabus RX, I continue receiving byte from the device after "stop echo UART RX". I try the flush method from the python serial library, but it does not work.

My output from my script : 

```
Switching to BBIO mode
Switching to UART mode
Starting baurate detection, turn on your serial device now
Press Ctrl+C to cancel
Stop echo UART RX
Byte receive after stop echo UART: b'\x01'
set baudrate ret=b'\x01'
Selected baudrate: 9600
Start echo UART RX
byte=b'\xd4'
Stop echo UART RX
Byte receive after stop echo UART: b'\x01'
set baudrate ret=b'\x01'
Selected baudrate: 19200
Start echo UART RX
byte=b'\x9e'
Stop echo UART RX
Byte receive after stop echo UART: b'!'
Cannot stop echo UART RX mode
```

Baldanos, 2019-07-06:
This might be caused by the Python serial library buffering some bytes before you send the stop RX command.

`serial.flush()` is used to clear the TX buffer and not the RX buffer, so I would suggest to send the stop command, then read the whole RX buffer and look for the last byte (should be `b'\x01'`) :

```
def set_baudrate(baudrate):
    # Deactivate echo UART RX
    print("Stop echo UART RX")
    serial.write(b'\x03')
    val = serial.read(serial.in_waiting)
    print(val)
    if b'\x01' not in val[-1]:
        print("Cannot stop echo UART RX mode")
        return False
```

Looking forward to seeing your script once its done.

bvernoux, 2019-07-13:
Could you provide a feedback with this issue ? (as for me it is solved in code now)

ghecko, 2019-07-14:
Yes, it work properly. I will share with you the standalone script when It will be clean. It work properly using BBIO_UART_BRIDGE and ask user to press UBTN to change baudrate. You can close this issue

bvernoux, 2019-07-14:
Thanks for feedback the script is welcome as Pull request in https://github.com/hydrabus/hydrafw/tree/master/contrib

closed 2019-07-14 by bvernoux