hydrabus/hydrafw #101: BBIO_UART_BRIDGE write does not work

ghecko, 2019-07-12:
When entering BBIO_UART_BRIDGE, it is not possible to send any byte to the target.

I use this script:

```
import serial

ser = serial.Serial('/dev/hydrabus', 115200)

for i in xrange(20):
    ser.write("\x00")
if "BBIO1" not in ser.read(5):
    print "Could not get into bbIO mode"
    quit()

print "Into BBIO mode"

print "Switching to UART mode"
ser.write('\x03')
if "ART1" not in  ser.read(4):
    print "Cannot set UART mode"
    quit()

print "Setting speed"
ser.write('\x6a')
if ser.read(1) != '\x01':
    print "Error setting speed"
    quit()

print "Starting BBIO_UART_BRIDGE"
ser.write('\x0F')
```

End then, connect to hydrabus using screen. Any user input is not sent to the target. However, I can properly see the output from the target device.

ghecko, 2019-07-13:
The problem is solved. Indeed, the issue is due to my target device. All work fine with another.
You can close this issue

bvernoux, 2019-07-13:
Thanks for the feedback

closed 2019-07-13 by bvernoux