hydrabus/hydrafw #102: I2c scanner : Wrong addresse shown ?

Jugulaire, 2019-08-23:
Hi, 

I'm happy with my new Hydrabus. Lot of protocol supported and such a nice documentation around. 
But today, i was working on i2c related stuff and got a weird bug. 
When i was running the scan command hydrabus returned `0x3c`. 

But after trying to send some data to this address i always got NACK. 
So, after some time passed to check everything on my wiring/parameters i decided to fire up my Bus Pirate to check. 

I've ran the scan function and get this : `Searching I2C address space. Found devices at:
0x78(0x3C W) 0x79(0x3C R)`

So i decided to look around to understand whats going on, why this address is totally wrong.. 
And finally i got the answer. 
I2C address is coded on 7 bit but it seem that HydraFW is showing in in 8bit. 

It can be pretty cool to modify this in future version to use similar output as BusPirate because it's more trivial.

bvernoux, 2019-08-23:
I confirm the address returned by HydraFW is an I2C 7bits address (and not 8bits) like in your example **0x3c** which is correct.
Note: **0x3c** is a 7bits value and to convert it to 8bits you shall do a left shift of 1 bits like that in C 
`0x3c << 1` (corresponding also to 0x3c * 2 if you prefer) which returns 0x78 for Write address (bit0 = 0) or 0x79 (bit0=1) for I2C Read address.

I confirm it will be better to provide details like on BusPirate including both 7bits address and 8bits address with R & W address for clarity purpose.

See code https://github.com/hydrabus/hydrafw/blob/master/src/hydrabus/hydrabus_mode_i2c.c#L314
Extract of actual code:
```
	/* Skip address 0x00 (general call) and >= 0x78 (10-bit address prefix) */
	found = FALSE;
	for (i = 0x1; i < 0x78; i++) {
		bsp_i2c_start(I2C_DEV_NUM);
		bsp_i2c_master_write_u8(I2C_DEV_NUM, i << 1, &ack);
		bsp_i2c_stop(I2C_DEV_NUM);
		if (ack) {
			cprintf(con, "Device found at address 0x%02x\r\n", i);
			found = TRUE;
		}
	}
```
I have updated the Wiki about that 
https://github.com/hydrabus/hydrafw/wiki/HydraFW-I2C-guide

Your feedback is welcome
My proposal is to also modify code of HydraFW to provide same result as BusPirate for clarity purpose (including both 7bits & 8bits addresses and clearify Read/Write values)

Jugulaire, 2019-08-24:
Hey, 

Many thanks for clarification. 
I gonna take good note of this, now i know where to read code related to each function in the FW if i got some questions.

bvernoux, 2019-09-01:
Proposal is to replace line https://github.com/hydrabus/hydrafw/blob/master/src/hydrabus/hydrabus_mode_i2c.c#L319:
`cprintf(con, "Device found at address 0x%02x\r\n", i);
`
by
`cprintf(con, "Device found at address 0x%02x (0x%02x W / 0x%02x R)\r\n", i, (i << 1), (i << 1)+1);
`

Baldanos, 2019-09-10:
Fix committed to master

closed 2019-09-10 by Baldanos