hydrabus/hydrafw #103: Publish pyHydrabus on PyPI

ghecko, 2019-08-29:
Hi,

It would be relevant to publish pyHydrabus on Pypi.

Baldanos, 2019-08-29:
Hi !

pyHydrabus is still under development and there might be possible API changes.
As soon as the module is stable enough (ie all the current features are tested) I'll publish it on pypi.

Feel free to comment this issue with things you've tested and/or want to see modified.

ghecko, 2019-09-04:
Hi,

I didn't see your response before.
Great, I will probably create some pull requests to add new features and change asserts' call (If no working issue is induced) as soon as possible, are you OK with this?

Baldanos, 2019-09-05:
Sure ! 
Feel free to send us pull requests.

Baldanos, 2019-12-28:
pyHydrabus has been published on Pypi : https://pypi.org/project/pyHydrabus/

closed 2019-12-28 by Baldanos