hydrabus/hydrafw #111: Scanning with HydraNFC seems to do nothing

pmathelin, 2019-11-14:
Hi,

I'm working on HydraBUS with HydraNFC and I can't get NFC scan to work.
I don't know if i'm doing something wrong. I just uploaded HydraFW v9 on the board via DFU and followed the steps desribed in the folowing guide: https://github.com/hydrabus/hydrafw/wiki/HydraFW-HYDRANFC-guide#read-uid-of-an-isoiec_14443-tag-only-type-a-limited-to-4bytes-uid

Nothing happend when I try to scan my NFC Tag. 
Type: ISO 14443-3A, NXP MIFARE Ultralight

So in the console, I did:
> nfc typea
NFC> scan

And also tryied:
NFC> scan continuous period 500 (or just continuous without specifying a period)

But in both cases nothing happend. I just got a message "Scanning MIFARE with 500ms period. Press user button to stop." when I specify. Otherwise, I just got nothing..

Do you have any idea of what am i doing wrong and how to fix it?

Thank you for your help!
BR,
Pierre

pmathelin, 2019-11-14:
I found the reason of my issue. It seems that the antenna frequency where it emit is affected by the tag that I used. After analysation, I fixed the issue by placing my tag at 3 centimeters from the antenna and not directly sticked. Now, the antenna is capable to scan my tag correctly and displays the results that I expected.

closed 2019-11-14 by pmathelin

bvernoux, 2019-11-14:
Thanks for the feedback this limitation is because TRF7970A chipset is not clever enough (even if it was the best when it was released) to manage correctly some tags when they are too close.
This issue will be solved with HydraNFC v2 which use a totally new chipset with some very advanced hardware/software features to do automatic tuning, a very good AGC (Automatic Gain Control) integrated and tons of other unique features ...