hydrabus/hydrafw #112: Issue with sniffer which seems to sniff wrong data

pmathelin, 2019-11-20:
Hi,

I am working on Hydrabus with hydra NFC and trying to use the sniffer mode.
I followed the documentation to start sniffing: https://github.com/hydrabus/hydrafw/wiki/HydraFW-HydraNFC-guide#autonomousstand-alone-sniffer-mode

My issue there, is that it seems that the data I sniff is not the correct one. Most of the time I have a write_file() error and sometimes I have write_file() OK but with some data from I don't know where.
I am using a tag iso 14443-3A MIFARE Ultralight and I have no problem with scaning this tag. This tag is not empty. There is a text "Hello world!" saved on it.

To do the sniffing, I putted the HydraNFC between my tag and the phone used as reader, by spacing them to each other. I don't find what am i doing wrong.
Can anyone help me with my issue please ? :)

Thank you in advance,
Pierre

bvernoux, 2019-11-20:
The important point for good sniffing is to be not too far (less than 2cm) and not too near(sometimes depending on reader/tag) between the tag & reader.
  * Be careful to do not touch components on the HydraNFC antenna (or have them in contact with something not isolated) during sniff session.
  * If you use UFL connector cable be careful that it is not rotating and short some components, I recommend to solder both SMA with a SMA female to female cable for best results and to avoid any potential issue because of UFL connector.

1) What is the firmware you are using ?
   * Thanks to use latest version of hydrafw v0.9+ see https://github.com/hydrabus/hydrafw/releases/tag/v0.9-beta

2) Could you provide which command you have entered for the sniff ?

3) Could you take a short video of your setup during sniffing and provide corresponding sniff results with what is expected/obtained on the reader ?

pmathelin, 2019-12-02:
Hi,
Thank you for your response. 
So, I tried to use an SMA instead of the UFL connector.
I still have some issue while sniffing using hydratool and `nfc sniff bin frame-time` command.

Hydratool response of the FTDI is 

> Frame size error resync data: 00 00 00 00

Do you have any idea of what am I doing wrong?
Following, you can find a screenshot of my windows:

[image]

The FTDI that I use is the same as in the tutorial. I plugged the yellow to PA9 and the black to GND.
The FW version is the last one (build_HydraFW_v0.9-beta-65-g55e5dbe_HydraBus_HydraNFC.zip).

Thank you for your help

bvernoux, 2019-12-02:
1) I suspect the issue could be due to your build of hydratool v0.3.2.0 built with Qt 5.12.6 see my build I use with MSVC2013 32bits and Qt 5.9.1 [hydratool_win32_Qt5_9_1.zip](https://github.com/hydrabus/hydrafw/files/3912376/hydratool_win32_Qt5_9_1.zip)

2) Are you using a native Windows OS (if yes which one ?) as FTDI High Speed under a VM does not work correctly (maybe because of USB HS emulation which is not fast enough ...) ?

pmathelin, 2019-12-03:
Thanks for your answer,

I am using a Windows 10 professional native OS.
I tried to use your build of hydratool and the firmware that you linked me to have the same basis as you. So i used putty for the commands and hydratool for te FTDI.
My tag is also a Mifare UL but my reader is a smartphone samsung which have an NFC reader/writer.

[image]

I still have the same issue, so maybe that's come from my reader?

bvernoux, 2019-12-03:
Thanks for the details

It will be interesting that you validate the sniffer work fine with a normal reader like SCL3711 and a MifareUL or MifareClassic Credit Card size ... (I use it for my own test and even with a small antenna the signal is very good and sniffing is very stable even on very long session and full dump without any error...)
Also thanks to check the size of your Tag as the best are Credit Card size Tag else it is harder on small tag especially with reader with limited antenna and power like Phone.

In your case the main issue is probably the phone NFC which is too far (and probably limited in power) or and without enough power (too small antenna) from sniffer, so it is very hard to sniff if there is not enough power.
Anyway it is very strange that you always have some "00 "as in case power is not good enough it shall show some invalid values like FF ... and also you shall see Reader Query (as the Reader Query are very strong with ASK 100%/Modified Miller) and potentially not TAG as the tag answer is the hardest to sniff (with ISO14443A) as the TAG signal is modulated using OOK Manchester (only on about 10% of signal on top/bottom) and so sensitivity/good signal is critical to correctly decode TAG Answer ...

In paste I have validated the HydraNFC sniffer work with phone but only on a Samsung Nexus S (but I cannot test anymore on this one as it does not power on anymore since few years ...) and only on few specific position directly on antenna with big tag credit card size...

You could change the RX gain for  your usage (with a low power field and small antenna like there is on phone) see file https://github.com/hydrabus/hydrafw/blob/master/src/hydranfc/hydranfc_cmd_sniff.c#L267 (By default it is **Gain reduction for 5 dB B2=1&B3=0**)
See datasheet https://www.ti.com/document-viewer/TRF7970A/datasheet/detailed-description#SLOS7431033 "RX Special Setting Register (0x0A)" B2 & B3 => Sets the RX gain reduction and reduces sensitivity,.
  * You can try "Gain reduction 0 dB" gd1=0, gd2=0 => **tmp_buf[1] = 0x31;**
  * Note1: B4 (Appropriate for highest bit rate (848 kbps) used in high-bit-rate ISO/IEC 14443) can be also set to 0 for test
  * Note2: Yes B0 shall be set to 1 and it is not documented but it corresponds to AGC no limit, you can also play with other bits to check if that improve or not the RX sensitivity ...

Sniffer sensitivity /range  is one the drawback of TRF7970A when used as passive sniffer (which is a mode not documented when used to sniff both Reader/TAG) and the AGC is not very good too (it is why I have decreased the Gain to 5dB to avoid saturation....) and sensitivity is not very good too.

A dedicated NFC Sniffer Antenna (like ISO 10373-6 / Calibration Coil tuned for sniffing purpose) should improve that as default HydraNFC Antenna is designed to be used as Active Reader and so when used as Passive sniffer (or also Card Emulator) it attenuate (detuning effect because of 3 antennas Reader/Sniffer/Tag) the sniffed signal which also explain that the sensitivity/range is very short (max 4cm in best conditions with Credit Card size Tag).

pmathelin, 2019-12-04:
First of all, thank you for your time.

I tried to do the same manipulation with another NFC reader. Unfortunately, nothing changes..
Just for your info and to check if I am doing it well, please find some media below:
[image]
[image]
[image]
[image]

I hope that my Tag is enough large.
I'll continue my researches and keep you in touch if I find anything.

Thank you for your help

bvernoux, 2019-12-04:
Thanks for the photos
* I confirm the C232HM USB 2.0 HS is correctly connected (PA9 & GND)
* It seems the capacitors on the HydraNFC Antenna are not the good one to be confirmed (as it is hard to see colors on photo) so that could explain it does not work
Could you confirm that when using the hydrafw nfc> scan on Mifare UL and MifareClassic all work fine ?
Could you also test with a Vicinity tag with hydrafw nfc> scan ?
As if I'm right and some capacitor are wrong on Antenna or even on the HydraNFC in that case hydrafw nfc scan shall not read correctly Tags especially at different distance to check sensitivity ...

Could you provide me (on my Email bvernoux@gmail.com) hi-definition photo (if possible with correct color with white balancing...) of top of HydraNFC and the same for HydraNFC Antenna ?
We will continue by Email as it seems you have something defective

bvernoux, 2020-01-29:
This issue seems to be related to FTDI interface when pushed to limit @8.4Mbauds
So far there is no workaround when FTDI does not work @8.4MBauds as anyway it is not designed to go so fast

closed 2020-01-29 by bvernoux