hydrabus/hydrafw #115: Can't control GPIO pins

YautjaRO, 2019-12-26:
In version 0.9A gpio works, in new compiled version under Windows i get error (system don't understand pin name):
`>` gpio mode out on
Please select at least one GPIO pin.
Command failed.
`>` gpio PA1 mode out on
Invalid command.
gpio PA1 mode out on
-----^
`>` gpio pb1 mode out on
Invalid command.
gpio pb1 mode out on
-----^
`>` gpio pb* mode out on
Invalid command.
gpio pb* mode out on
-----^
`>` show system
HydraFW (HydraBus) v0.9-beta-95-g4462884 2019-12-18
sysTime: 0x009d0b09.
cyclecounter: 0x41f45fda cycles.
cyclecounter64: 0x0000000241f45fe9 cycles.
10ms delay: 1680031 cycles.

MCU Info
DBGMCU_IDCODE:0x10076413
CPUID:        0x410FC241
Flash UID:    0x490021 0x4D4D5017 0x2037394E
Flash Size:   1024KB

Kernel:       ChibiOS 5.1.0
Compiler:     GCC 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
Architecture: ARMv7E-M
Core Variant: Cortex-M4F
Port Info:    Advanced kernel mode
Platform:     STM32F405 High Performance with DSP and FPU
Board:        HydraBus 1.0
Build time:   Dec 26 2019 - 20:17:17

bvernoux, 2019-12-27:
In fact it is an issue with documentation and it is because we are using an updated version of tokeline (see https://github.com/hydrabus/tokenline) and the old "free form string" now requires quotes.
So now for any **<string>** parameter you shall add quotes at start of string and at end of string.
`gpio PA1 mode out on `
shall be replaced by
`gpio "PA1" mode out on`

It seems the only impact in documentation is towards [HydraFW-GPIO-guide](https://github.com/hydrabus/hydrafw/wiki/HydraFW-GPIO-guide) so it is a good point to let this issue open until we release the Firmware v1.0 and fix the documentation towards that evolution.

This will be fixed in wiki when we will publish the v1.0 Firmware soon (as today the documentation is towards latest stable released and not towards branch: master)

bvernoux, 2019-12-27:
I have updated https://github.com/hydrabus/hydrafw/wiki/HydraFW-GPIO-guide to add this information
I have tested each example and I confirm they work like expected

YautjaRO, 2019-12-27:
I check writing pin names with quotes - yes, it works!
Thanks a lot for replay and solution!

closed 2019-12-27 by YautjaRO