hydrabus/hydrafw #119: STM32CubeIDE project

ddavidebor, 2020-04-08:
Hello.

It would be really nice if you could bootstrap a STM32CubeIDE project.

If you don't know what it is, basically ST has acquired a company called Atollic that made an IDE called Atollic Truestudio, that was kinda the best IDE for STM32.

I have made a couple of tries but performing this first bootstrapping requires someone more experienced with this project. 

I'm an electronic engineer, i wanted to verify if it were possible to add ethernet to the project, being that the microcontroller supports it, but it's not easy to do such things without a well configured project.

bvernoux, 2020-04-10:
The AIM of HydraFW is to have an easy/intuitive and fully configurable multi-tool in order to interact with external world (other MCU/Peripherals...)
1) You can build HydraFW with STM32CubeIDE just click on "Import ..." and choose "General->Existing Projects into Workspace" then choose "hydrafw" from src directory and click on build (it will requires you have installed GCC ARM for your platform see https://github.com/hydrabus/hydrafw/wiki/how-to-build-flash-and-use-hydrafw-on-windows or https://github.com/hydrabus/hydrafw/wiki/Getting-Started-with-HydraBus-flash-and-use-hydrafw-on-linux)
2) We are already using STM32Cube/STM32 Drivers see https://github.com/hydrabus/hydrafw/tree/master/src/drv/stm32cube/stm32f4xx_hal (potentially later we could add other STM32 Drivers depending on new features)
I know it is not a bootstrap (and so we are not using features of STM32CubeIDE especially configuration of MCU pins/Driver graphically...) because we need our own abstraction layer (HAL in root of stm32cube directory with files prefixed by bsp) and we need to reconfigure in "live" (depending on HydraBus Mode) each peripheral/IO so it does not really match with a "static peripheral/IO code" like what is done with GUI of STM32CubeIDE.
The interesting concept of STM32CubeIDE is to do a configuration from GUI to initialize/configure the peripherals/IO to a default mode as today we do that from scratch using ChibiOS and also STM32Cube Drivers.
3) You are welcome if you have any idea to improve actual architecture/code... (for Information I often use STM32CubeIDE GUI to map/check the MCU peripherals/IO mainly for HW design to have maximum flexibility for routing & features...)

ddavidebor, 2020-04-12:
Thank you for the info, I will close the issue for now to avoid clutter.

closed 2020-04-12 by ddavidebor