hydrabus/hydrafw #125: Build of Hydrafw fails

M-a-t-h-i-R, 2020-11-11:
Hi,
I am trying to flash the newest Firmware 0.10 on to the Hydra Bus. I failed to accomplish my goal via the guide (https://github.com/hydrabus/hydrafw/wiki/Getting-Started-with-HydraBus-flash-and-use-hydrafw-on-windows). As an alternative I tried to flash via the Cube IDE. I followed https://github.com/hydrabus/hydrafw/wiki/Getting-Started-with-HydraBus-and-STM32CubeIDE for this option. This options fails for me too.
When Building the Project (Step 6) the Build fails.
The Error I received from the Build Console is:
`10:52:09 **** Incremental Build of configuration Debug for project hydrafw ****`
`make -j4 all `
`make: Makefile:15: pipe: No such file or directory`
`Compiling crt0_v7m.S`
`Creating ./common/hydrafw_version.hdr`
`Compiling vectors.S`
`Compiling chcoreasm_v7m.S`
`make (e=3): Das System kann den angegebenen Pfad nicht finden.`
`make: *** [Makefile:323: common/hydrafw_version.hdr] Error 3`
`make: *** Waiting for unfinished jobs....`
`"make -j4 all" terminated with exit code 2. Build might be incomplete.`

`10:52:10 Build Failed. 2 errors, 1 warnings. (took 708ms)`

In the c File common a File is called with `#include "hydrafw_version.hdr` but there is no such file in the repositories. 

I am probably the problem, but I would appreciate any help.

Baldanos, 2020-11-11:
Hi !

Looks like the `hydrrafw_version.hdr` cannot be generated correctly.
From the `src/` folder, are you able to run the following command ?
```
python build-scripts/hydrafw-revision.py
```

This should return the current git version number (`0.10` in this case).

If that's not the case, please follow the steps indicated here : https://github.com/hydrabus/hydrafw/wiki/how-to-build-flash-and-use-hydrafw-on-windows#prerequisites-for-windows-install-git-python-and-clone-hydrafw

Let me know if you have any other issues.

Baldanos, 2020-11-20:
Closing since it's been more than a week with no update

closed 2020-11-20 by Baldanos

M-a-t-h-i-R, 2020-11-20:
> It seems the error you have is due because you have not followed the steps and you have not cloned the repository and probably just extracted the zip archive and so the python build-scripts/hydrafw-revision.py cannot work as it does not find git information...
> Please confirm what steps you have done.

Yesterday I was able to flash the device successfully after reinstalling all tools. It seems that I failed at structuring the Data in the right way. So my guess is that your assumption is correct.

M-a-t-h-i-R, 2020-11-20:
I would like to apologize for the long latency period and thank you for the help.

Baldanos, 2020-11-20:
No worries ;) Glad you were able to flash the device.