hydrabus/hydrafw #128: How to use Direct Mode 0

blackwiz4rd, 2021-01-26:
Hi,
I'd like to use hydranfc to read the raw signal using the direct mode 0 of the TRF7970A chip. I want to obtain a raw signal such as the two signals (rx and tx) at page 46 of the chip's manual: https://www.ti.com/lit/ds/slos743m/slos743m.pdf?ts=1611650098010. Those signals look to be captured with an oscilloscope.
I have seen that you suggested in the `CHANGELOG` to read the signal with a logic analyzer on PC2 from PIN 23 of the chip (PB1 in hydrabus): 

> Added dm0 command => Direct Mode 0 Sniffer Test work fine mainly for test with a Logic Analyzer on PC2

Would it be possible to obtain those signals with a cheap usb digital analyzer (e.g. https://www.amazon.it/AZDelivery-Logic-Analyzer-compatibile-version/dp/B01MUFRHQ2/ref=sr_1_1?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=usb+digital+analyzer&qid=1611666497&sr=8-1)?

bvernoux, 2021-01-27:
Yes you can use such cheap LogicAnalyzer connected on HydraBus PC2 pin (which corresponds to TRF7970A IO6)
For more details see also https://github.com/hydrabus/hydranfc/blob/master/hardware/HydraNFC_modes_and_pinout.pdf
Note :The best/simpler way with advantage to see both Reader & Tag is to use the sniffer Mode with Infinite Trace see https://github.com/hydrabus/hydrafw/wiki/HydraFW-HydraNFC-v1-guide#sniffer-iso14443a-with-unique-hard-real-time-infinite-trace-mode

blackwiz4rd, 2021-01-27:
Hi, thanks for answering my question. As I have never used a logic analyzer before I do understand that it reads the input pin data at a certain sampling rate (e.g. 24MHz) and transforms the incoming data into a signal made of bits. However, I would like to obtain the most raw signal as possible as in the figure from the manual which seems to be in volts. Is an oscilloscope the only way to do so?
Thanks.

bvernoux, 2021-01-28:
Direct Mode 0 Sniffer Test does not output analog values as it output digital values so only 0 or 3.3v (it is why a logic analyzer can be used to capture that)
If you want to capture analog signal you need an oscilloscope with a specific Coil (usually Calibration Coil ISO 10373-6) see my tutorial / PDF for different steps to decode NFC Vicinity 15693 (but we can apply the same process for ISO 14443-A/B to capture the analog signal)
https://hydrabus.com/NFC_ISO_IEC_15693_Vicinity_HydraNFC_read_UID_PicoScope_PulseView_BVERNOUX_23Nov2019.pdf

blackwiz4rd, 2021-02-05:
> Direct Mode 0 Sniffer Test does not output analog values as it output digital values so only 0 or 3.3v (it is why a logic analyzer can be used to capture that)
> If you want to capture analog signal [...]

Thank you very much.

closed 2021-02-05 by blackwiz4rd