hydrabus/hydrafw #130: How to keep gpio (PIN) state when leave and reenter BBIO PIN mode ?

mengguang, 2021-07-04:
Everytime HydraBus enter the BBIO PIN mode, the gpio will be re initialized.    
https://github.com/hydrabus/hydrafw/blob/32d53fccb5352b246aea4d71cf46990a83b7bc16/src/hydrabus/hydrabus_bbio_pin.c#L50
How to keep gpio (PIN) state when leave and reenter BBIO PIN mode ? 
The use case is common. 
For example: the ssd1306 display in SPI mode, we need to drive a D/C pin to choose send data or command over SPI.   
What do you suggest ?

bvernoux, 2021-07-04:
A simple way for your specific use case is to change int32_t pin_mode[8];	uint32_t pin_pull[8]; as global static initialized to MODE_CONFIG_DEV_GPIO_IN and MODE_CONFIG_DEV_GPIO_NOPULL that shall do the trick to keep old value when you enter/exit/enter again in PIN mode the same type of trick shall be done in other used mode(using same GPIO) as they will restore default state on enter or exit...
Note: It is not planned to change actual code as each mode is intended to reinit GPIO as safe state (Input) by default, potentially we could add a configuration settings (a file in sdcard for example with a special value) to change that to be discussed with @Baldanos

mengguang, 2021-07-05:
Thanks for your kindly suggestion.  
I will try to fork the firmware and make it work.  
Also I think this is a common demand for most application that need to drive at least two peripheral.

closed 2021-07-05 by mengguang

Baldanos, 2021-07-06:
You can also use [auxiliary pins](https://github.com/hydrabus/hydrafw/wiki/Auxiliary-pins) (up to 4) in any BBIO mode. The configuration for these pins does not reset if you change the mode during the BBIO session.

Using [pyHydrabus](https://github.com/hydrabus/pyHydrabus), it's quite easy: 

```
>>> s = pyHydrabus.SPI()
>>> s.AUX[0].direction = pyHydrabus.OUTPUT
>>> s.AUX[0].value=1
```

mengguang, 2021-07-06:
@Baldanos Thanks. This is exactly what I need. 4 pins is enough. Very good!