hydrabus/hydrafw #147: CAN BUS FILTER

Alfa16bravo, 2023-04-04:
hello, after several tries when I use the filter for id we can filter only one id, and when I use the mask example filter mask 0x208, the only data received is with the sid: 0x72 why? even by changing the mask the only result obtained is 0x72. and sometimes the continuous command crashes the hydrabus. thanks

Alfa16bravo, 2023-04-04:
I tried to apply an id filter in version v0.10 and it works. v0.11 not working

bvernoux, 2023-04-05:
Could you explain why do you have closed the issue ? (do you have solved it with latest firmware v0.11)
Maybe the filter was not set/applied correctly ? (to be checked with @Baldanos)

Alfa16bravo, 2023-04-05:
hello, yes I bought hydralincan a smillier, when I apply the id 0x208 filter only, nothing changes I have to apply the 0x07ff mask and it works the first 2 frames that are displayed are generally not filtered but the rest yes, sometimes the hydrabus crashes in the can mode, the adj can wheels on the hydralincan allow you to adjust what? THANKS

bvernoux, 2023-04-05:
> hello, yes I bought hydralincan a smillier, when I apply the id 0x208 filter only, nothing changes I have to apply the 0x07ff mask and it works the first 2 frames that are displayed are generally [...]

We will check that with @Baldanos 
- If you need more details on the hardware HydraLINCAN you shall contact smillier / check also https://github.com/smillier/HydraLINCAN

bvernoux, 2023-11-27:
Checked with @Baldanos lot of time ago and this issue cannot be reproduced, so you have probably done a mistake for the filter
If you can provide more details to reproduce the issue with the exact use case

bvernoux, 2024-01-09:
No feedback since lot of time issue closed 
If you can reproduce the issue and provide more details with full details how to reproduce it, feel free to create a new issue

closed 2024-01-09 by bvernoux