hydrabus/hydrafw #164: Add a read with timeout feature

Baldanos, 2024-02-13:
#160 started to implement a *read with timeout* feature for smartcard mode.

This feature seems very interesting, but would require additional work in order to be applicable to all modes.

As of now, timed out read does not show any value, and this change would allow to print the received bytes.

```
uart1> 0x41 r:2 0x42
WRITE: 0x41
READ error:3
WRITE: 0x42
uart1> 
```

Q: How to implement this ? Adding a new `tread` command in CLI like in #160 or modify the `read` command altogether ?

bvernoux, 2024-02-13:
Good question I think the most transparent will be to modify read command and add a parameter to configure the timeout of read command

Baldanos, 2024-02-23:
PR #165 opened

closed 2024-03-30 by Baldanos