hydrabus/hydrafw #171: UART Bridge Mode Issue: Fails to Open Serial Port in WebSerial

seuros, 2024-05-01:
I am encountering an issue with the Hydrabus device when trying to use UART bridge mode. Specifically, I am unable to connect to the device using WebSerial via a browser. The connection attempt fails with the error "Failed to open Serial port."

**Steps to Reproduce**

- Connect the Hydrabus device to the computer.
- Switch the Hydrabus to UART bridge mode using the appropriate commands. 
  - uart 1
  - speed 115200
  - bridge
- Open a browser and navigate to a WebSerial application. example https://web.esphome.io
- Attempt to connect to the Hydrabus device.

**Expected Behavior**
I expect the browser to successfully connect to the Hydrabus device in UART bridge mode without any errors.

**Actual Behavior**
The browser fails to connect to the Hydrabus device, and I receive an error message stating "Failed to open Serial port."

Board: Hydrabus v1.0

0x8008135, 2024-05-01:
Hello,

Looks like you have a permission issue. I tested and could get it listed and even interact with the shell or uart bridge.

[image]

bvernoux, 2024-05-01:
Yes I confirm I have tested on my side with Windows it works perfectly using Chrome browser (or any browser which support correctly WebSerial)

seuros, 2024-05-01:
I confirm, Windows worked flawlessly.
 
For Linux i had to add the user to the dialout group, then reset the session.

```
$ sudo usermod -aG dialout $USER
```

Thanks