hydrabus/hydrafw #174: Proposal: Provide the Binary as a github artefact.

seuros, 2024-05-06:
I'm thinking that the repo should provide the compiled binary once a version released.

I see that the majority of the people will use the generic codebase unless they are contributors or building something on top.

bvernoux, 2024-05-07:
Yes it is a good idea but before to release an official version compiled binary, so far the preferred way is to do that manually to carefully build and test it on our own computer (so far there is no any automatic test to do the tests) and so we are 100% confident the official binary released is the one we have tested (as we could always have issue on the built version done on GitHub).
Any idea feedback is welcome to be checked with @Baldanos

bvernoux, 2026-01-16:
You are welcome to open a PR we have already the build it is just not released automatically today ...