hydrabus/hydrafw #175: gcc-arm-none-eabi upgrade ?

seuros, 2024-05-07:
Is there a reason why gcc-arm-none-eabi is kept locked to a 10-year-old version? 

I upgraded it locally with good success, however i didn't try all the protocols.

bvernoux, 2024-05-07:
The reason is all was validated with that old version and I'm 100% sure everything work fine (without potential regression like we had on lot of GCC ARM version):
GCC ARM blocker issues we had for reference:
- Links:
  - https://github.com/hydrabus/hydrafw/wiki/how-to-build-flash-and-use-hydrafw-on-windows
  - https://github.com/hydrabus/hydrafw/wiki/how-to-build-flash-and-use-hydrafw-on-linux
- Blocker issue:
  - "WARNING: GCC ARM 4.8.x & [GCC ARM 5.0/5-2016-q1](https://bugs.launchpad.net/gcc-arm-embedded/+bug/1570613) up to GCC ARM 8-2019-q3-update (https://bugs.launchpad.net/gcc-arm-embedded/+bug/1570613) shall be avoided as they crash during link of chibios ..." 
  - Those issues have took so much time to be fixed/merged officially in at end of 2021 by GCC ARM Dev Team...
     - https://bugs.launchpad.net/gcc-arm-embedded/+bug/1570613

The other reason is newer gcc-arm does not bring anything in performance/size for that project.

Anyway if we want to switch to newer gcc-arm version it will requires to test everything (there is lot of things which requires lot of hardware) and so far we do not have such setup (all in one to validate everything at once) and even less in an automatized way to do it.

seuros, 2024-05-07:
Thank you for your response.

closed 2024-05-07 by seuros